Xinyuan-LilyGO/T-PicoC3 #1: Communicating Between Processors

MadGeometer, 2022-05-14:
Hello,

Can you provide some ideas on getting the processors work together?

- Do they share any memory or LittleFS/SPIFFS/VFS
- Is it possible to communicate between them using TX and RX, or some other way?
- Do they have any pins in common?

Thanks,
Mike

LilyGO, 2022-05-25:
They are separate chips..Our design can communicate via TX and RX or SPI. The pin can see the details page

closed 2022-05-27 by LilyGO

lesept777, 2022-09-02:
Hi
I'm trying to set communication between the 2 processors.
As only the RP2040 seems to be able to control the red led and the display, I use it as receiver.

On the ESP32C3 side, I run the following code (called com_RP.py):
```
import machine
from time import sleep

RP2040 = machine.UART(1, baudrate=115200, tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11), txbuf=1024, rxbuf=1024)
RP2040.init()
print('ok')
sleep(10)
for i in range (15):
    RP2040.write(str(i))
    print(i)
    sleep(1)
```

It waits 10 seconds, then sends a number each seconds 15 times. I expect it to run at boot by using the boot.py file, which contains the line:
`import com_RP`

On the RP2040 side, I do the same: boot.py contains the line
`import com_ESP32`
and the file com_ESP32.py contains the following code:
```
from machine import UART, Pin
from time import sleep
import st7789
import tft_config
import NotoSans_32 as font

led = Pin(25, Pin.OUT)
for i in range(10):
    led.value(not led.value())
    sleep(0.05)
led.off()
tft = tft_config.config(1)
tft.init()
bg = st7789.BLUE
fg = st7789.WHITE
tft.fill(bg)
tft.write(font, "waiting...", 0, 20, fg, bg)
print ("waiting...")
ESP32 = machine.UART(1, baudrate=115200, tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11), txbuf=1024, rxbuf=1024)
ESP32.init()

while True:
    if ESP32.any() > 0:
        line = ESP32.readline()
        print("*"+str(line)+"*")
        if (line):
            led.value(not led.value())
            tft.fill(bg)
            tft.write(font, line, 0, 20, fg, bg)

```

What is expected is: It blinks the led, inits the display (blue background and message) then inits the uart and waits for a message. When a message arrives, it displays it on the display.

When I power the device on, the  com_ESP32.py runs (led is blinking and display is blue with the message), but nothing seems to arrive from the UART. Maybe the com_RP.py does not run on the ESP32C3 side, I have no way to verify. Or the UART is not properly configured.

Can anyone explain or help me? Thanks.