Xinyuan-LilyGO/T-PicoC3 #4: Request for Examples in Micropython

mperino, 2022-05-25:
Can you provide some examples in Micropython that show using the wifi, and the LCD display?

Poedersuiker, 2022-05-25:
I am trying to get the LCD working with Micropython and machine.SoftSPI. But would love some examples as well, as I still have a blank screen.

Bradovman, 2022-05-27:
I am not using the default machine.SoftSPI and am instead using this https://github.com/russhughes/st7789_mpy custom build of micro python. However you could try setting phase to 1 in machine.SoftSPI, as that seemed to work for me.

Poedersuiker, 2022-05-27:
> You can refer to this project, the LCD pins are the same for both of them https://github.com/Xinyuan-LilyGO/LILYGO-T-display-RP2040/tree/main/example/MicroPython/TFT Do you need me to integrate the project?

A simple link to the firmware and hello world script would help very much. Especially with the pins and steps to get everything displayed.

mperino, 2022-05-27:
I'd like to re-open the issue.

Please provide some examples in this repo that demonstrate the use of:
1. The Wifi
2. LCD Display

Some super simple example that at least gives us the module, and pins used.. 
For example I updated the Adafruit Circuit Python ESP_AT examples for another board:

https://github.com/mperino/Adafruit_CircuitPython_ESP_ATcontrol/blob/main/examples/esp_atcontrol_aio_post.py

The request is a simple example in MicroPython for the two main features of this board.  

Thank you.

Poedersuiker, 2022-05-28:
Screen stays blank:

I used this firmware [T-DISPLAY-RP2040](https://github.com/russhughes/st7789_mpy/tree/b4aa060b74e2d2490770fa92310571f7602059f9/firmware/T-DISPLAY-RP2040)
And this tft-config.py [tft-config.py](https://github.com/russhughes/st7789_mpy/blob/b4aa060b74e2d2490770fa92310571f7602059f9/examples/configs/tdisplay_rp2040/tft_config.py)

trying to display the hello.py example.

LilyGO, 2022-05-31:
I will update the LCD with WIFI connection example shortly, the update is done with MicroPython and I will reply the link here

mperino, 2022-06-05:
With the firmware from:
https://github.com/russhughes/st7789_mpy/tree/b4aa060b74e2d2490770fa92310571f7602059f9/firmware/RP2

And the diagram on the main page, I'm not sure which is the SPI pin.
Here's the code:
`import machine
import st7789
spi = machine.SPI(4, baudrate=40000000, polarity=1, sck=machine.Pin(2), mosi=machine.Pin(3))
display = st7789.ST7789(spi, 135, 240, reset=machine.Pin(0, machine.Pin.OUT), dc=machine.Pin(1, machine.Pin.OUT))
display.init()`

and it gives the error:
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
ValueError: SPI(5) doesn't exist

I've tried pin's 4 and 5 for the SPI but neither seems to work.

mperino, 2022-06-06:
Sucess!

The tft_config_py needs to have phase = 1!:

> def config(rotation=0, buffer_size=0, options=0):
> 
>     Pin(22, Pin.OUT, value=1)
> 
>     spi = SPI(0,
>         baudrate=62500000,
>         polarity=1,
>         **phase=1,**
>         sck=Pin(2, Pin.OUT),
>         mosi=Pin(3, Pin.OUT),
>         miso=None)

Then do a reset of the board.
I'll do a quick youtube of how to get a board displaying the example text and make some smaller simpler demos as well. 
Still at zero progress on the Wifi.  Any hints, tips, or clues appreciated.

LilyGO, 2022-06-06:
I have integrated the ST7789 driver in this warehouse and will update it later
https://github.com/Xinyuan-LilyGO/lilygo-micropython

LilyGO, 2022-06-06:
Using WIFI, you can try to communicate with ESP32-C3. The pin definitions are described in detail in the schematic diagram on the home page. The esp-at directive is used by default. You can refer to this documentation on how to drive it https://docs.espressif.com/projects/esp-at/en/latest/esp32/index.html

mperino, 2022-06-06:
Is there a recommended micropython library on the Pico side like the ones for circuitpython?
Under CircuitPython there are helper libraries adafruit_espatcontrol, and adafruit_espatcontrol_wifimanager that allow the Pico to talk to the ESP-32 as a wifi co processor..

Poedersuiker, 2022-06-06:
> Sucess!
> 
> The tft_config_py needs to have phase = 1!:
> 
> > def config(rotation=0, buffer_size=0, options=0):
> > ```
> > Pin(22, Pin.OUT, value=1)
> > 
> > spi = SPI(0,
> >     baudrate=62500000,
> >     polarity=1,
> >     phase=1,
> >     sck=Pin(2, Pin.OUT),
> >     mosi=Pin(3, Pin.OUT),
> >     miso=None)
> > ```
> 
> Then do a reset of the board using the on-board reset switch. I'll do a quick youtube of how to get a board displaying the example text and make some smaller simpler demos as well. Still at zero progress on the Wifi. Any hints, tips, or clues appreciated.

That worked, Great work

ExperiMentor, 2022-06-07:
Thanks mperino - that gets it working.
I found it hard to identify the repo directories within the video, so have written out the instructions for reference.
There was one 'error' - it is not necessary to copy the font across - the device already contains all the fonts used in the example programs. Several of the other examples also work without additional tweaks and show off the display nicely

1.	Boot the LilyGo-TpicoC3 whilst pressing BOOTSEL button. From ‘firmware\T-DISPLAY-RP2040’ drag firmware.uf2 to the RPI-RP2 drive using File Explorer. 
NOTE that this IS NOT THE SAME as the standard firmware that would be loaded by Thonny.
2.	From ‘examples/configs/tdisplay_rp2040’ , copy tft_config.py to the RP2040.
Open this file in Thonny. Edit the line ‘phase = 0’ to be ‘phase = 1’. Save.
3.	From ‘examples’, copy any or all of the following to the RP2040 and run: 
feathers.py, hershey.py, noto_fonts.py, scroll.py, hello.py

jimemo, 2022-11-12:
Adding my experience with the display to this thread. 
It took me a while to get the display working on the T-PicoC3 because the suggested russhughes library linked from this project has different versions. Specifically, this file from this project:
https://github.com/Xinyuan-LilyGO/T-PicoC3/blob/main/example/Micropython/lcd_config/tft_config.py
...doesn't work with the russhughes library also linked from this project:
https://github.com/russhughes/st7789_mpy/tree/b2e790191a4833b5ef15ab0a851c0337967c8840
Here is what worked for me in 5 steps:
**1.** Install Raspberry Pico firmware in Thonny
**2.** Get the library file from https://github.com/russhughes/**st7789py_mpy**/blob/master/lib/st7789py.py
**3.** For the font I used a bitmap from a different russhughes project: https://github.com/russhughes/st7789_mpy/blob/b2e790191a4833b5ef15ab0a851c0337967c8840/fonts/bitmap/vga1_8x16.py
**4.** Save the following code as tft_config.py and include it in your main.py:
```
from machine import Pin, SoftSPI
from time import sleep
import st7789py as st7789

def config():

    Pin(22, Pin.OUT, value=1) 

    spi = SoftSPI(
        baudrate=20000000,
        polarity=1,
        phase=0,
        sck=Pin(2),
        mosi=Pin(3),
        miso=Pin(13))

    return st7789.ST7789(
        spi,
        135,
        240,
        reset=Pin(0, Pin.OUT),
        cs=Pin(5, Pin.OUT),
        dc=Pin(1, Pin.OUT),
        backlight=Pin(4, Pin.OUT),
        rotation=1)
```
**5.** Finally, here is my main.py program to display text on 4 lines in 4 different colors:
```
import st7789py as st7789
import tft_config
import vga1_8x16 as font

tft = tft_config.config()
tft.fill(0) #clears after previous display

def displayTxt(L1,L2,L3,L4):
    x = chr(32)
    #uses end space to clear the end of the line each time
    endSpace = str(x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x+x)
    if L1:
        tft.text(font,L1+endSpace,15,15,st7789.GREEN,st7789.BLACK)
    if L2:        
        tft.text(font,L2+endSpace,15,40,st7789.BLUE,st7789.BLACK)
    if L3:        
        tft.text(font,L3+endSpace,15,65,st7789.GREEN,st7789.BLACK)
    if L4: 
        tft.text(font,L4+endSpace,15,90,st7789.RED,st7789.BLACK)
        
if __name__=='__main__':
        displayTxt("Hello Line 1","Hello Line 2","Hello Line 3","Hello Line 4")
```

jimemo, 2022-11-12:
@mperino here is a solution for connecting to wifi with example code:
**1.** ESP library, I tried loads and settled on this one, it has some bugs but it does mostly what I need:
https://github.com/heggiedavid/PIco-ESP01-MP/blob/main/esp.py
**2.** My program **main.py** file:
`import time
from esp import ESP

uart_id=1
tx_pin=8
rx_pin=9
baud_rate=115200
tx_buffer=2048
rx_buffer=2048

esp = ESP(uart_id,tx_pin,rx_pin,baud_rate,tx_buffer,rx_buffer,debug=True)
secrets = {"ssid":"MY-SSID","password":"MY-PASSWORD"}

#wifi for esp co-processor devices
def wifiConnect():
    print('SSID: '+secrets['ssid'])
    print('Password: '+secrets['password'])
    try:
        esp.soft_reset()
        esp.send_at_cmd("AT+CWMODE=3\r\n", timeout=3)
        time.sleep(2)
    except:
        return False
    if not esp.is_connected:
        try:
            AP = esp.remote_AP
            if AP[0] != secrets["ssid"]:
                for i in range(3):
                    reply = esp.send_at_cmd('AT+CWJAP="' + secrets['ssid'] + '","' + secrets['password'] + '"', timeout=10, retries=3)
                    if (b"WIFI CONNECTED" in reply) or (b"WIFI GOT IP" in reply):
                        print("ESP wifi connected.")
                        return True
                    else:
                        print("ESP wifi retry "+str(i))
                        continue
                print("ESP wifi fail after retries.")
                return False
        except:
            return False

if __name__=='__main__':
    while(1):
        if wifiConnect():
            print('Wifi connected.\r\n')
            break
        else:
            print('Wifi retry...\r\n')
            continue
`

psousa2021, 2022-11-14:
I'm a beginner and I'm having difficulties because I don't have the examples with Display and with Wi-fi too, can someone help me please

hajulied, 2022-12-25:
@jimemo 
Hi,
which software has to be installed on the esp32 part?

hajulied, 2022-12-26:
@jimemo:
I installed factory_Mini-1.bin on the esp32.
Changed in  esp.py the following values (similar to the main.py):
def __init__(self,
                 uart_id=1,  #0
                 tx_pin=8,    #0
                 rx_pin=9,   #1
                 baud_rate=115200,
                 tx_buffer=2048,  #1024
                 rx_buffer=2048,
                 debug=False

an now it works.
Now I'm looking for a solution to transmit some sensor-values via WLAN to a MQTT-server on a RaspberryPi.

mmMicky, 2022-12-27:
> I'm a beginner and I'm having difficulties because I don't have the examples with Display and with Wi-fi too, can someone help me please
> 
> I tried to use the examples given above but there seems to be a lack of code in the lib esp

I suggest you read the factory example, there will be the simplest way to use tft and wifi. You should create an issue.

jimemo, 2023-01-11:
> @jimemo Hi, which software has to be installed on the esp32 part?

I didn't change it, I used the factory esp32. You should be able to find it in the repository.

hajulied, 2023-01-11:
Thank you,  I forgot to edit my Post. I used it also in the descripton of my solution.