Xinyuan-LilyGO/T-PicoC3 #5: How to Install Drive under MacOS

DawnMagnet, 2022-05-30:


LilyGO, 2022-05-31:
It is divided into two drivers, one is the driver of C3. The other one is the driver of RP2040. You can choose to install according to your needs, select the current communication chip through the USB2.0 positive and negative cable connection

mybigman, 2022-06-06:
Same issue... the RP2040 works fine but am unable to see the C3.

Does it need a USB to TTL adapter, if not what chipset does it use to be able to install the driver?

DawnMagnet, 2022-06-06:
> Same issue... the RP2040 works fine but am unable to see the C3.
> 
> Does it need a USB to TTL adapter, if not what chipset does it use to be able to install the driver?

I thought maybe RP2040 works under USB mode, so every platform supporting USB will works fine. But ESP32C3, I've already installed espressif's tool(esptool), but esptool can't find the chip. maybe this specified chip change something? I couldn't tell.

LilyGO, 2022-06-11:
> Same issue... the RP2040 works fine but am unable to see the C3.
> 
> Does it need a USB to TTL adapter, if not what chipset does it use to be able to install the driver?

It can be connected to ESP32C3 by speaking the USB reverse connection. It communicates with the computer by connecting the USB pins. ESP32C3 can update programs directly via USB without serial chip.

LilyGO, 2022-06-11:
> I thought maybe RP2040 works under USB mode, so every platform supporting USB will works fine. But ESP32C3, I've already installed espressif's tool(esptool), but esptool can't find the chip. maybe [...]

If you connect the ESP32C3 correctly, the serial port of the computer will display a serial port named (USB JTAG/serial debug unit)

LilyGO, 2022-08-02:
Looks like the problem has been resolved

closed 2022-08-02 by LilyGO