Xinyuan-LilyGO/T-PicoC3 #6: Request for ESP32C3 routines in Micropython

ExperiMentor, 2022-06-11:
In another issue, you wrote:
`Using WIFI, you can try to communicate with ESP32-C3. The pin definitions are described in detail in the schematic diagram on the home page. The esp-at directive is used by default. You can refer to this documentation on how to drive it https://docs.espressif.com/projects/esp-at/en/latest/esp32/index.html`

But I and others cannot get it to work. It should be something like the program below, but whatever I write to the UART, the response is just my write echoed back. I never get OK or any other text.

I'm using the LilyGO Micropython firmware and have not changed the ESP32C3 firmware from factory (indeed I cannot find any way to change it - booting with the USB-C plug inverted gives green light, but with BOOTSEL pressed, I don't see any drive in Windows).

What is wrong?

My attempts:
from machine import UART, Pin
import utime

# Try finding a power pin to turn ESP32 on  - pin 22? THIS MAKES NO DIFFERENCE
ESP32C3power = machine.Pin(22, machine.Pin.OUT)
ESP32C3power.high()

ESP32 = machine.UART(1, tx=Pin(8), rx=Pin(9)) #, cts=Pin(10), rts=Pin(11)) # leave cts and rts undefined to mirror the c example
ESP32.init(baudrate=115200, bits=8, parity=None, stop=1, flow=0) # Flow can be 0 (not None), UART.RTS, UART.CTS or UART.RTS | UART.CTS)

# Read before writing
while True:
    #Recvd = ESP32.readline() # read a line # or try with .read(100)
    Recvd = ESP32.read()
    if Recvd == None:
        print("No data read before writing")
        break
    if len(Recvd) == 0:
        break
    print(len(Recvd), "characters:", Recvd)

# write command
ESP32.write("AT+GMR\n")

while ESP32.any() == 0: # wait until some data available
    utime.sleep(0.01) # 0.001 is too fast; gets 10 charaters rather than whole line

# Read & print until data flow stops
while True:
    Recvd = ESP32.read() # all available chars; or .read(100) or .readline()
    if Recvd == None:
        print("No data read this loop")
        break
    if len(Recvd) == 0:
        print("Length or received data is zero (this does not seem to happen; get Recvd == None instead)")
        break
    print(len(Recvd), "characters:", Recvd)
    utime.sleep(0.1)

print("No more data coming back")

ExperiMentor, 2022-06-13:
I'm now making progress - the (first) error was that the AT commands need to be followed by CR LF ("Windows style") and not just LF ("Linux / Unix style"). So changing the write command from 
ESP32.write("AT+GMR\n")
to 
ESP32.write("AT+GMR\r\n")
gets responses from the ESP32C3
That command gives me a response of "ERROR" but at least it's progress!

Sending "AT\r\n" gives a total of 3 lines of response:
Received from ESP32C3 4 chars: b'AT\r\n'
Received from ESP32C3 2 chars: b'\r\n'
Received from ESP32C3 4 chars: b'OK\r\n'

Omitting the echo of the command and blank lines, trying to connect to my WiFi:
AT+CWMODE=1\r\n gets "OK"
AT+CWJAP="ssidname","password"\r\n gets "WIFI CONNECTED", "+CWJAP:4", "WIFI DISCONNECT", "ERROR"

But at least making progress ....

LilyGO, 2022-06-13:
ESP32C3 can modify the firmware. I will introduce to you.
There are three LED lights next to the USB. One is a PICO programmable LED. The other two are corresponding to ESP32C3 TO USB and PICO TO USB indicators. The purpose of their existence is to tell us which chip is currently connected to the USB (there may be a PICO light on, but the ESP32C3 is connected. The reason is that the wiring method of the USB 2.0 cable cannot be the same)

The programming method of ESP32C3: first short-circuit the IO9 pin and GND (note that the label is IO9). Then connect to the USB (note that the esp32c3 needs to be connected. If there is no word like USB JTAG/serial debug unit in the device manager or serial communication tool, please connect the USB in reverse). Then use ESP_TOOL tool to burn

mperino, 2022-06-15:
@LilyGO for the esp_at_port.py what is the bus_num?
If you could provide a simple example using esp_at_port.py, that would be appreciated.

ExperiMentor, 2022-06-16:
Sending GMR **too fast** gets the response "busy", not sending it **too many times**.

This is my testing program:

from machine import UART, Pin
from utime import sleep

# setup
ESP32uart = UART(1, baudrate=115200, tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11))

ESP32uart.write('ATE0\r\n') #turn off echo; ATE1 turns echo back on if desired
        
def sendAT(cmd): 
    if len(cmd) != 0: # add AT and \r\n unless empty command sent
        cmd = 'AT'+cmd+'\r\n'
    else:
        cmd = '\r\n'
    
    ESP32uart.write(cmd)

    while ESP32uart.any() == 0: # loop quickly until start of response
        sleep(0.01)

    sleep(0.1) # Wait until even a long response is completely received  
    return ESP32uart.read() # Note that blank lines etc have NOT been stripped out

try:
    while True:
        command = input("Command: ")
        response = sendAT(command)
        # print(response, "\n") # if want to see all the \r\n
        print(response.decode('UTF-8')) # otherwise display with codes applied

except KeyboardInterrupt:
    pass

# Examples to try. Don't include the AT+ or \r\n
# +GMR               Give Info about the ESP32 device
# +CWMODE=1          1 sets station (client) mode; 2 is Access Point (server); 3 is both together
# +CWJAP="ssidname","password"   Join Access Point
# +CWLAP             List Access Points
# +CWQAP             Quit Access Point ie disconnect
# +CIPSTA?           Gets info about IP address, gateway, subnet; ERROR if not connected
# E0                 Echo off; 1 is On
# There are commands to send X bytes of data, or to act as passthrough where it relays everything until +++

LilyGO, 2022-08-02:
Looks like the problem has been resolved

closed 2022-08-02 by LilyGO