Xinyuan-LilyGO/T-PicoC3 #7: Help with bricked on the rp2040 side

unokits, 2022-06-24:
I had installed different MicroPython firmwares to try different projects, and then I decided to come back to the LilyGO firmware from this repo. After installing it (see URL below, _Latest commit [8999f3d]_), I can no longer start the filesystem by holding BOOT and inserting USB cable. I've also ready the README.md and tried different variations of hold BOOT, click RESET, release BOOT after 1 second. I've also tried that sequence while inserting USB cable. It appears to be bricked on the Pico side.

https://github.com/Xinyuan-LilyGO/T-PicoC3/blob/main/example/Micropython/firmware/firmware.uf2

ExperiMentor, 2022-06-24:
Please clarify what you mean by "start the filesystem". There are only 2
processes:
1. To load a new firmware, hold BOOTSEL whilst connecting power. It matters
which way up you put the USB plug as the other way up is for loading
firmware to the ESP32. You want the way up showing blue LED. If you see the
other colour (green I think?) Its wrong way up.
2. Connect power without holding a button: runs main.py if present and lets
you program using Thonny etc

LilyGO, 2022-08-02:
Looks like the problem has been resolved

closed 2022-08-02 by LilyGO