Xinyuan-LilyGO/T-PicoC3 #9: WIFI & Bluetooth Accessible on through ESP32 mode ?

ianteohsc, 2022-07-09:
Noob question. Is the WIFI and Bluetooth functionality when the device is powered in ESP32 mode (ie green light) ? If not, where can i find example code to access WIFI and Bluetooth using Micropython on RP2040 firmware ?

ianteohsc, 2022-07-09:
I found the solution and am able to get WIFI working using the russhugh firmware v1.19.1-18-g6e868d47d-dirty

Use the esp_uart class provided here: https://github.com/Xinyuan-LilyGO/T-PicoC3/blob/main/example/Micropython/esp32c3_port/esp_at_port.py

ESP32 commands can be referenced here:
https://docs.espressif.com/projects/esp-at/en/latest/esp32

closed 2022-07-09 by ianteohsc

ExperiMentor, 2022-07-09:
This topic was covered more thoroughly in Issue #6.
We are now able to communicate with the ESP32 using microphone, but there
does not seem to be a full library of user-friendly routines.