Xinyuan-LilyGO/T-PicoC3 #10: Flashing latest ESP AT firmware

alexphobby, 2022-08-01:
Please help us with a guide on how to flash latest ESP32 C3 AT Firmware
I did tried with ESP32-C3-MINI-1-V2.3.0.0\factory\factory_MINI-1.bin @ 0 without success
with usb plugged so i get green led on

On pressing start button, nothing happens, remains IDLE. Also could not connect with telnet on COM port

LilyGO, 2022-08-02:
It has been described in the introduction if you are burning the C3, so I don't know which step you stopped at now.https://github.com/Xinyuan-LilyGO/T-PicoC3/blob/main/README.md

alexphobby, 2022-08-02:
I did try to burn the firmware using the espressif windows tool (in this repo I did find 2 screenshots with the settings)

alexphobby, 2022-08-02:
Sorry, closed by mistake

LilyGO, 2022-08-02:
Have you tried this step?

>Method 2: (if there is no serial port when connecting ESP32C3 USB) disconnect USB, short connect ESP32C3-IO9 and GND, and then reconnect USB (note that ESP32C3 device is selected to connect USB).

lesept777, 2022-08-31:
Hi, I have the same problem. When trying to connect with Thonny, and flash the firmware (this one : https://github.com/russhughes/st7789_mpy/tree/master/firmware/T-DISPLAY-ESP32 or this one: https://micropython.org/resources/firmware/esp32c3-20220618-v1.19.1.bin ???) I only get this error message:
```
Unable to connect to COM13: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Un périphérique attaché au système ne fonctionne pas correctement.', None, 31)

If you have serial connection to the device from another program, then disconnect it there first.

Process ended with exit code 1.
```
Using Windows, the ESP32 appears on the device manager as USB Port 13. I'd like to try your Method 2:

> Method 2: (if there is no serial port when connecting ESP32C3 USB) disconnect USB, short connect ESP32C3-IO9 and GND, and then reconnect USB (note that ESP32C3 device is selected to connect USB).

How is it working? 

1. short IO9 and GND
2. connect USB
3. release IO9
Is this correct? Then what?

lesept777, 2022-09-02:
Thank you for your answer.
I did the process (short IO9 to GND, connect USB, release IO9) and I could access the flashing process in Thonny. I flashed the firmware found here (https://github.com/russhughes/st7789_mpy/tree/master/firmware/T-DISPLAY-ESP32) but now the device keeps booting on and on and I get this error message in Thonny: 

> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> invalid header: 0xffffffff
> PROBLEM IN THONNY'S BACK-END: Internal error (serial.serialutil.SerialTimeoutException: Write timeout).
> See Thonny's backend.log for more info.
> You may need to press "Stop/Restart" or hard-reset your MicroPython device and try again.
> 
> 
> Process ended with exit code 0.
If I try to flash another firmware (https://micropython.org/resources/firmware/esp32c3-20220618-v1.19.1.bin) specific for ESP32C3, I get the same result.
What is wrong? How can I get the ESP32 to work properly?
Thanks for your help

LilyGO, 2022-09-02:
T-PicoC3 has no ESP32 chip on board, only ESP32-C3 and rp2040. If you need to burn mpy firmware to C3. Your firmware needs to choose ESP32C3 or generic_c3

lesept777, 2022-09-02:
Can you please provide a link to the appropriate firmware?

lesept777, 2022-09-02:
I found it ! https://micropython.org/download/esp32c3-usb/
Thanks for your help!

To summarize, here is what I did: 
1. short IO9 to GND, connect USB, release IO9
2. erase flash, unplug USB
3. short IO9 to GND, connect USB, release IO9
4. flash firmware, unplug USB
5. launch Thonny

To erase flash, I opened a powershell or command window, then typed:
`_C:\Path\To\Thonny_\python.exe -u -m esptool --chip esp32c3 --port COM13 erase_flash`
To flash the firmware:
`_C:\Path\To\Thonny_\python.exe -u -m esptool --chip esp32c3 --port COM13 --baud 460800 write_flash -z 0x0 _C:\Path\To\Firmware_\esp32c3-usb-20220618-v1.19.1.bin`

Hope this will help someone...

closed 2022-09-05 by LilyGO