Xinyuan-LilyGO/T-PicoC3 #11: Tear free display.

davidnr, 2022-08-02:
Is it possible to generate smooth animation on the display ? As far as I can see ST7789 Vsync pin is not connected and there isn't either a MISO so we can't get vtrace information.

LilyGO, 2022-08-03:
Increase the SPI rate, when the clock rate reaches 32M (135x240x16x60), it can reach 60HZ.
Actually image processing and register setting take some time.

davidnr, 2022-08-03:
That won't fix tearing, it will just provide (potentially) more FPS, but I want to avoid having images cut in half because I'm updating the ram while a retrace is being done. This is usually usually handled by waiting for vsync signal before writing to vram , in order to avoid memory being overwritten while being read (thus, causing tearing).

[image]
.

LilyGO, 2022-08-04:
In which scene are you using the screen tearing. Most of the ST7789's screen modules do not have access to the Vsync pin.

closed 2022-09-05 by LilyGO