Xinyuan-LilyGO/T-PicoC3 #13: Address red led from ESP32C3

lesept777, 2022-09-02:
How to address the led using the ESP32C3? I tried this code:
```
from time import sleep
from machine import Pin

led = Pin(25, Pin.OUT)
while True:
    led.on()
    sleep(1)
    led.off()
    sleep(1)
```

But I get the following error message, as if 25 is not allocated as a pin number: 

> Traceback (most recent call last):
>   File "<stdin>", line 4, in <module>
> ValueError: invalid pin

This exact same code works correctly from the RP2040.
Is this due to the firmware I flashed ? (https://micropython.org/download/esp32c3-usb/ : ESP32-C3 with USB, matest)

lesept777, 2022-09-04:
I have seen [here ](https://github.com/micropython/micropython/blob/d2e4cf00ccec432ccd38adea78099887ac578882/ports/esp32/machine_pin.c#L120)that the ESP32C3 can only address GPIOs 0 to 13, 20 and 21. Which one is connected to the red led?

LilyGO, 2022-09-05:
The ESP32-C3 is not connected to any of the LEDs, so you can't light the LEDs.

lesept777, 2022-09-05:
OK, thank you

closed 2022-09-05 by lesept777