Xinyuan-LilyGO/T-PicoC3 #14: Communicating Between Processors

lesept777, 2022-09-02:
Hi
I'm trying to set communication between the 2 processors.
As only the RP2040 seems to be able to control the red led and the display, I use it as receiver.

On the ESP32C3 side, I run the following code (called com_RP.py):
```
import machine
from time import sleep

RP2040 = machine.UART(1, baudrate=115200, tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11), txbuf=1024, rxbuf=1024)
RP2040.init()
print('ok')
sleep(10)
for i in range (15):
    RP2040.write(str(i))
    print(i)
    sleep(1)
```

It waits 10 seconds, then sends a number each seconds 15 times. I expect it to run at boot by using the boot.py file, which contains the line:
`import com_RP`

On the RP2040 side, I do the same: boot.py contains the line
`import com_ESP32`
and the file com_ESP32.py contains the following code:
```
from machine import UART, Pin
from time import sleep
import st7789
import tft_config
import NotoSans_32 as font

led = Pin(25, Pin.OUT)
for i in range(10):
    led.value(not led.value())
    sleep(0.05)
led.off()
tft = tft_config.config(1)
tft.init()
bg = st7789.BLUE
fg = st7789.WHITE
tft.fill(bg)
tft.write(font, "waiting...", 0, 20, fg, bg)
print ("waiting...")
ESP32 = machine.UART(1, baudrate=115200, tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11), txbuf=1024, rxbuf=1024)
ESP32.init()

while True:
    if ESP32.any() > 0:
        line = ESP32.readline()
        print("*"+str(line)+"*")
        if (line):
            led.value(not led.value())
            tft.fill(bg)
            tft.write(font, line, 0, 20, fg, bg)

```

What is expected is: It blinks the led, inits the display (blue background and message) then inits the uart and waits for a message. When a message arrives, it displays it on the display.

When I power the device on, the  com_ESP32.py runs (led is blinking and display is blue with the message), but nothing seems to arrive from the UART. Maybe the com_RP.py does not run on the ESP32C3 side, I have no way to verify. Or the UART is not properly configured.

Can anyone explain or help me? Thanks.

_Originally posted by @lesept777 in https://github.com/Xinyuan-LilyGO/T-PicoC3/issues/1#issuecomment-1235640189_

lesept777, 2022-09-03:
Maybe I should send a endofline character as follows?
`RP2040.write(str(i)+'\n')`
I'll try this very soon

LilyGO, 2022-09-05:
It is recommended that you refer to the schematic diagram to configure the communication pins.

lesept777, 2022-09-05:
What do you mean?

ExperiMentor, 2022-09-05:
No, the schematic tells you the logical port pin numbers for the connections that are already wired

lesept777, 2022-09-05:
Thank you for your answer. But I don't understand. I only see UART0 RX, TX on pins 12,13 and 16,17 on the RP2040 side, but nothing on the ESP32. Do I just need to use GPIO12 and 13, as:
`RP2040 = UART(1, baudrate=9600, tx=13, rx=12) # ESP32 side`
and
`ESP32 = UART(1, baudrate=9600, tx=Pin(12), rx=Pin(13)) #RP2040 side`
... this is not clear...
Even if I try: `RP2040 = UART(1)` and `ESP32 = UART(1)` they don't seem to communicate.

Or should I wire the pins I want to use? For example, if I define on the ESP32 side:
`RP2040 = UART(1, baudrate=9600, tx=9, rx=8)`
and on the RP2040 side:
`ESP32 = UART(1, baudrate=9600, tx=Pin(28), rx=Pin(29))`
then I should add a wire between GPIO8 and GPIO 28, and another one between GPIO9 and GPIO29.
Is this correct?

LilyGO, 2022-09-06:
[image]
[image]
[image]

My screenshots indicate the pins where the two chips are connected. IO8 IO9 of RP2040 and IO6, IO7 of ESP32-C3.

lesept777, 2022-09-07:
For those who would like to use the UART communication between the two processors, here is what I did:

ESP32 side:
```
from machine import UART
RP2040 = UART(1, baudrate=38400, tx=7, rx=6, cts=5, rts=4)
```

RP2040 side:
```
from machine import UART
ESP32 = UART(1, baudrate=38400, tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11))
```

To read data from the UART:
```
 if ESP32.any() > 0:
    buf = ESP32.readline()
    line += buf.decode("utf-8")
```

It seems that higher baudrates do not work.
Hope this will help anyone.

closed 2022-09-07 by lesept777

ExperiMentor, 2022-10-11:
According to the schematic (
https://github.com/Xinyuan-LilyGO/T-PicoC3/blob/main/Schematic/T-PicoC3.pdf),
you should be using:
On the RP2040: TX, RX, CTS and RTS (physical chip pins 11, 12, 13 & 14)
(logical ports GPIO8, GPIO9, GPIO10 & GPIO11)
On the ESP32-C3: RX, TX, RTS & CTS (physical chip pins 13, 12, 10 & 9)
(logical ports MTDO, MTCK, MTDI, MTMS). I don't understand the logical port
naming on the ESP32-C3 but believe they are multifunctional.

These physical chip pins are not brought out to board pins, so you cannot
wire to them; they are connected already - presumably correctly with TX to
RX etc