Xinyuan-LilyGO/T-PicoC3 #15: Factory example not compiling

anders0l, 2022-09-03:
Hello there!

Provided library in /example/factory/pico/firmware has got lot's of errors:
- wrong function location - Use of undeclared identifier errors
- setTX, setRX - No member named 'setRX' in 'arduino::UART'
- no printf in tft lib - No member named 'printf' in 'TFT_eSPI'

even there is an issue in pico library - missing stdio.h in packages/arduino/hardware/mbed_rp2040/3.2.0/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/

Did you try to compile provided library on your own?
If yes, please try to recompile it one more time and if everything fine, please provide your environment and which libraries (with versions) do you use.

And I'm not alone - https://steinlaus.de/ttgo-t-display-rp2040-hello-world/

LilyGO, 2022-09-05:
[image]
I use this library.

anders0l, 2022-09-05:
Thank's for your reply!

Ahh, okay, make sense.
Kindly ask you - please add description about it somewhere in the docs, as you use non official library.

It is https://github.com/earlephilhower/arduino-pico

For platformio I've tried https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst
After setup, there were no issues in main file (except functions order)
But there we some issues in libraries, and with fast fixes, it's finally compiled (haven't tried it yet on t-pico)

Thank you!

LilyGO, 2022-09-06:
OK, I'll mark it up in the documentation.

closed 2022-09-06 by LilyGO

Luciano2018, 2022-10-05:
Which TFT-eSPi version need to use? The last get error of compiling.

LilyGO, 2022-10-06:
> Which TFT-eSPi version need to use? The last get error of compiling.

As far as I know any version can be supported. Please post your error warning log.

Luciano2018, 2022-10-06:
Thanks for answer, finally i've could compile sucessfully the factory firmware example, with all libraries in latest version and the board rp2040 in the last version too, instead of 1.13 seen in the issues. I get success choosing a different setup in User_setup.h, using this `#include <User_Setups/Setup137_LilyGo_TDisplay_RP2040.h>  // Setup file for Lilygo T-Display RP2040 (ST7789 on SPI bus with 135x240 TFT)` instead of this `#include <User_Setups/Setup135_ST7789.h>           // Setup file for ESP8266 and ST7789 135 x 240 TFT` 

### Resume:
Board LiLYGO T-PicoC3 T-Display
Arduino IDE 2.0 on Windows 11 22H2
File: `firmware.ino` from factory example. In `pin_config.h` can edit wifi credentials `const char ssid[] = "Your-WiFi";`
`const char pass[] = "Your-WiFi-Password";`

Board **Raspberry Pi Pico/RP2040** version 2.6.0 by Earle F.Philhower, III
Library **WiFiEspAT** version 1.4.1 by Juraj Andrassy
Library **OneButton** version 2.0.4 by Matthias Hertel
Library **TFT_eSPI** version 2.4.72 by Bodmer
Edit file `User_Setup_Select.h` and comment with `//` the line `#include <User_Setup.h>           // Default setup is root library folder` and uncomment deleting `//` to the line `//#include <User_Setups/Setup137_LilyGo_TDisplay_RP2040.h>  // Setup file for Lilygo T-Display RP2040 (ST7789 on SPI bus with 135x240 TFT)`

Thanks LILYGO!

HotNoob, 2022-11-26:
https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst should be pinned for platformio instructions.
would have saved such a headache if i found this ticket sooner.

jimemo, 2022-11-27:
@HotNoob PIO supports the T-Pico using the standard Arduino-mbed core that's built in. Here is my build script header from PIO:
```
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed @ 3.1.1
 - tool-openocd-raspberrypi @ 2.1100.0 (11.0)
 - tool-rp2040tools @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
```
To work with the ESPC3 for Wifi I am using this library (installed by PIO): https://github.com/khoih-prog/ESP_AT_Lib
Note that there is a workaround required to access Serial2 and TX-8, RX-9 on the T-Pico device. It's explained here: https://github.com/khoih-prog/ESP_AT_Lib/discussions/4
Here is my platformio.ini file that got generated from PIO GUI:
```
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
lib_deps = 
	khoih-prog/ESP_AT_Lib@^1.4.1
```