Xinyuan-LilyGO/T-PicoC3 #17: Blue LED?

lesept777, 2022-09-23:
There is a blue led on the back side of the board, opposite to the 3 leds from the USB C connector. What does it do, and is it addressable?

LilyGO, 2022-09-24:
Only one is controllable by the RP2040. The other two LEDs are indicator lights, which are used to indicate which chip is currently communicating with the USB.

lesept777, 2022-09-24:
Yes, I know. I was speaking about another LED, which is on the other side of the battery connector.
When you look at the back side of the board with the USB connector down, the 3 LEDs you mention are on the right. There is another blue LED on the left, close to the voltage regulator.
Is it addressable? What is its purpose?

ExperiMentor, 2022-09-25:
The schematic shows only three LEDs, only one of which is controllable (red, GPIO25).

lesept777, 2022-09-26:
[image]

LilyGO, 2022-09-26:
This is the battery charging indicator. Steady on during charging, and off when charging is complete or not charging. Blinks when no battery is connected.

lesept777, 2022-09-26:
Thank you. Does this mean that the board has a charging circuit, to charge a battery when plugged with the USB C connector?

ExperiMentor, 2022-09-26:
It seems that it does charge whilst running off USB-C power. I get blue LED whilst charging, off when a fully-charged battery is connected or if no battery connected. I've never seen flashing

[image]

lesept777, 2022-09-26:
Interesting, thanks! What is the size of the battery connector?

ExperiMentor, 2022-09-26:
It's a tiny 2-pin JST connector, smaller than the other ones I have. I got one connector with the 2 short wires supplied in the bottom of the plastic box that the LilyGo T-Pico-C3 came in.

FabriceAuz, 2022-10-29:
Thanks, you can close the issue

closed 2022-10-31 by LilyGO