Xinyuan-LilyGO/T-PicoC3 #19: Can both MCUs work at the same time?

Lwao, 2022-09-25:
While doing my research about the board it became unclear to me if both RP-Pico and ESP32-C3 can be used integrated at the same time running their own code and, perhaps, communicating between them. 

I understand the trick behind programming it with the USB-C cable, but I have not found anything about them working together and communicating to distribute the processing workload.

ExperiMentor, 2022-09-25:
Yes, it is normal and usual that both processors are powered from the same source, work at the same time and communicate with each other.
Reversing the USB plug only affects which chip is talking to the USB wire (and is particularly used for controlling which chip is being programmed).
Otherwise, the RP2040 is connected to the display, and the ESP32-C3 is connected to the WiFi antenna. Typically, you would program the RP2040 to do the main work and the ESP32-C3 to look after the WiFi connectivity - but you don't have to do exactly that. 
The two processors communicate with each other using UARTs which are hard-wired on the board (not external pins). There is discussion of that communication (trying toget it to work) in Issue 14 at https://github.com/Xinyuan-LilyGO/T-PicoC3/issues/14

Lwao, 2022-09-25:
That's awesome! Since now is clear, this feature reinforces my decision on acquiring this board. Thank you.

closed 2022-09-25 by Lwao