Xinyuan-LilyGO/T-PicoC3 #22: Need C++ example for ESP via serial/uart AT commands

jimemo, 2022-11-06:
I have this device working well with Micropython on the RP2040. It sends and receives to the internet via WiFi from the ESP chip on Serial/UART using AT commands. Also the LCD displays nicely with wifi status and readings from sensors connected. Sensor readings are uploaded to my server in JSON format. Can also receive JSON settings from the server.

Now I want to do the same but with C++ and I can't find examples for this device. I already tried Arduino-Pico and Earl Philhower libraries in VSCode / platformIO. 

RP2040 can't find/won't connect to ESP chip using hardware serial connection. It might be because of incorrect settings.

Please provide examples.

Thanks!

CelloGit, 2022-11-07:
Compliments on getting the RP2040 on this device to work with the display in MicroPython.

I never managed to, whatever advice I followed on this github such as settijng Phase to 1 in st7789_mpy-master\examples\configs\tdisplay_rp2040\tft_config.py.

Driving the display under Arduino is no problem.

I would really appreciate it if you could publish a small MicroPython code example that shows anything on the screen.

thanks

ExperiMentor, 2022-11-07:
In Issue #4, if you look at the end of the thread, it eventually gets to a working example in micropython. 

Please could you share your C++ or Arduino code examples as we are struggling with that too!

CelloGit, 2022-11-07:
"In Issue https://github.com/Xinyuan-LilyGO/T-PicoC3/issues/4, if you look at the end of the thread, it eventually gets to a working example in micropython."...

I could not get these to work.

CelloGit, 2022-11-07:
Working Arduino code:
a. Make a copy in a separate subfolder of the 3 files in Lilygo_T-PicoC3-main\example\factory\pico\firmware and rename firmware.ino to test.ino
b. In pin_config.h change "Your ssid" to the name of your own sssid, in between quotes
c. In pin_config.h change "password" to your wifi password, in between quotes
d. In test.ino, in lines 179 to 182, rename www.baidu.com to the website you want to show on the display.

jimemo, 2022-11-12:
@CelloGit I left my display solution on #https://github.com/Xinyuan-LilyGO/T-PicoC3/issues/4
Enjoy.