Xinyuan-LilyGO/T-PicoC3 #23: Pico example "firmware" won't compile

bratoff, 2022-11-13:
Installed TFT_eSPI, WiFiEspAT and OneButton using latest versions in Arduino Library Manager.  Commented out User_Setup.h and uncommented Setup135_ST7789.h in the User_Setup_Select.h as instructed in the firmware.ino source code comments.  Uncommented TFT pin definitions in pin_config.h and verified that they match the info for the T-PicoC3, as instructed in the source code.  Edited ssid and pass in pin_config.h to match my network.

When compiling, I get the following error:

> "C:\\Users\\bruce\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/python3" -I "C:\\Users\\bruce\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\2.6.3/tools/signing.py" --mode header --publickey "D:\\bruce\\Documents\\Arduino\\ESP32\\T-PicoC3\\firmware/public.key" --out "C:\\Users\\bruce\\AppData\\Local\\Temp\\arduino_build_832494/core/Updater_Signing.h"
> 
> Fatal Python error: initfsencoding: unable to load the file system codec
> ModuleNotFoundError: No module named 'encodings'
> 
> Current thread 0x00004c44 (most recent call first):
> 
> Using library TFT_eSPI at version 2.4.79 in folder: D:\bruce\Documents\Arduino\libraries\TFT_eSPI 
> Using library SPI at version 1.0 in folder: C:\Users\bruce\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.6.3\libraries\SPI 
> Using library LittleFS at version 0.1.0 in folder: C:\Users\bruce\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\2.6.3\libraries\LittleFS 
> Using library WiFiEspAT at version 1.4.1 in folder: D:\bruce\Documents\Arduino\libraries\WiFiEspAT 
> Using library OneButton at version 2.0.4 in folder: D:\bruce\Documents\Arduino\libraries\OneButton 
> exit status -1073740791> Error compiling for board Raspberry Pi Pico.
> 

I have tried compiling using both the "No FS" and the 64k FS memory models and I get the same compile error.

There is no customized TFT_eSPI in this repository and the only information given in the source code is to select the setup file in the User_Setup_Select folder.

What other changes are needed to make this code compile and run on the T-PicoC3?

bratoff, 2022-11-13:
After further investigation, ALL the Pico examples fail with the same error.

mmMicky, 2022-11-14:
You try to compile the most basic examples first, such as flashing lights. Also the driver library used in my example is an unofficial driver library.
https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

bratoff, 2022-11-14:
> You try to compile the most basic examples first, such as flashing lights. Also the driver library used in my example is an unofficial driver library. https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

If you read my comments, you'll see that I did try the other, simpler examples.  As it happens, the problem turned out to be a corrupted download of the RP2040 core, so I'm closing this issue as resolved.

closed 2022-11-14 by bratoff