Xinyuan-LilyGO/T-PicoC3 #24: PlatformIO - options for T-Pico

jimemo, 2022-11-27:
I spent a lot of time getting this set up and I'm sharing what I found. There are two ways to get the T-Pico (or any other Raspberry Pico board) working on PIO.

**Option 1:** earlephilhower arduino-pico core. Great if you're using Arduino but not fully supported (yet) on PlatformIO. There are some additional instructions to get it to work in the meantime: https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst 
There are plans to merge it into the main PlatformIO eventually: https://github.com/platformio/platform-raspberrypi/pull/36

**Option 2:** using the standard Arduino-mbed core that's built in to PIO. The packages and toolchain are all installed by the PIO GUI when you create a new project. Use these options in setup:
[image]
Here is my platformio.ini file that got generated from PIO GUI:
```
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
lib_deps = 
	khoih-prog/ESP_AT_Lib@^1.4.1
```
To connect to the ESPC3 via UART for Wifi I am using this library (installed by PIO GUI): https://github.com/khoih-prog/ESP_AT_Lib
Note that there is a workaround required to access Serial2 and TX-8, RX-9 on the T-Pico device. It's explained here: https://github.com/khoih-prog/ESP_AT_Lib/discussions/4
Here are the resulting build headers:
```
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed @ 3.1.1
 - tool-openocd-raspberrypi @ 2.1100.0 (11.0)
 - tool-rp2040tools @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
```

mmMicky, 2022-12-12:
It is an excellent introductory programming tutorial. Can you authorize me to put this on the readme page.

jimemo, 2022-12-12:
Yes go ahead @mmMicky . You can tag me in it.

jimemo, 2023-01-11:
@mmMicky I've updated the instructions, please update the readme file.

mmMicky, 2023-01-12:
@jimemo I modified a small part of it according to my understanding and updated to readme. Thanks again for your contribution to T-PICOC3.