Xinyuan-LilyGO/T-PicoC3 #29: What battery to buy?

mmendescortes, 2022-12-15:
I can only find 3.7v batteries, what are the specifications of the battery I need in order to use with the board?

mmMicky, 2022-12-15:
Please upload a picture of the battery, I need to judge.

rocketinventor, 2023-01-04:
@mmMicky How about this one?

[Lithium Ion Polymer Battery - 3.7v 2500mAh](https://www.adafruit.com/product/328)

[image]

mmMicky, 2023-01-05:
> @mmMicky How about this one?
> 
> [Lithium Ion Polymer Battery - 3.7v 2500mAh](https://www.adafruit.com/product/328)
> 
> [image]

This battery can be applied. But the interface doesn't seem to correspond. There should be a 1.25 pitch interface.