Xinyuan-LilyGO/T-PicoC3 #31: Comms via UART between RP2040 and ESP

alexphobby, 2022-12-16:
Hello
T-PicoC3 Tutorial for comms between internal RP2040 and ESP

RP2040(sender):

from machine import UART, Pin
import time
time.sleep(3)
uart = UART(1, baudrate=115200 , tx=Pin(8), rx=Pin(9), cts=Pin(10), rts=Pin(11))
uart.write('ATE0\r\n')
led = machine.Pin(25,Pin.OUT) #on board LED

while True:
    try:
        led.toggle()
        uart.write('TEST\r\n')
        print("uart OK")
    except:
        print("err")
    time.sleep(1)

ESP32(receier):

from machine import UART
uart = UART(1, baudrate=115000 , tx=7, rx=6, cts=5, rts=4)
uart.init(115000, bits=8, parity=None, stop=1)
uart.any() # >0 means data received
uart.read() #data being received

So, one use case scenario is to use ESP32 to send and receive messages from a MQ (HiveMQ tested by me).

Have fun with this nice board!

maxun, 2023-05-15:
Thank you Alex,

With your example, I was able to do something similar between RP2040 and ESP with arduino IDE.

If it can help someone else:

RP2040:
void setup()
{
  Serial2.setTX(8);
  Serial2.setRX(9);
  Serial2.begin(115200);
}

ESP32:
void setup() {
  Serial0.begin(115200);
  Serial0.setPins(6, 7, 5, 4);
}

Arquedy, 2023-06-01:
you two helped me a lot, i really appreciate it