Xinyuan-LilyGO/T-PicoC3 #32: RP2040 looses Micropython after plugoff

hajulied, 2022-12-18:
In Boot-Mode I can install the  firmware.uf2 via drag and drop. After copying the file, it is possible to reach the rp2040 with Thonny and try some examples. After plug-off and  plug.in the usb it is not possible to reach the RP. The USB-port is not availible. It seems , that RP "forgets" micropython?
What is my mistake? Or is RP damaged?

mmMicky, 2022-12-19:
It means that only the initial upload of mpy firmware can be connected normally. Unable to connect after reset? Does clicking the Thonny reconnect button also not work?

hajulied, 2022-12-19:
Installing Firmware. Uf2 is only possible in bootmode. The other Option via thonny is not possible, because there is no USB-Port. Trying the other way in thonny "install Firmware..." is not possible, because the answer is that the validity of the certicate has ended.
As long as i don't plug off the RP.. I See a com-Port. Pluggoff and plug on : no com Port?

hajulied, 2022-12-19:
... now I plugged the rp to power supply 5 volt, first nothing happens. After pressing the "run" button about 60 to 100 times (!) the main.py program starts. Thats strange.

mmMicky, 2022-12-20:
If it still exists, then the product may be defective.

hajulied, 2022-12-20:
I installed the factory firmware:
for RP.. :Xinyuan-LilyGO/T-PicoC3/tree/main/example/factory)/pico/
for ESP: ...factory/factory_MINI-1.bin
After plugoff/plugin the board starts with a Raspberry Pi Logo and switches to a screen, searching availible WLANs, trying to connect with a dummy key.
Windows shows a new Com-Port, but thonny can't connect to the board:
Unable to connect to COM4: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Ein an das System angeschlossenes Gerät funktioniert nicht.', None, 31)
If you have serial connection to the device from another program, then disconnect it there first.
Process ended with exit code 1.
How can I access the board?

hajulied, 2022-12-20:
booting in RP Mode shows:
Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython / firmware;
  - make sure the device is not in bootloader mode.

mmMicky, 2022-12-21:
Thonny should only be able to read the serial port information, but because the factory firmware is not mpy. So it reported an error I think it should be normal.
What needs to be tested now is whether it is normal every time it is powered on in the Arduino C environment. In addition, the mpy firmware has been made in our factory firmware. I tested it and it works fine.

mmMicky, 2022-12-21:
The usb port number simulated by mpy firmware is inconsistent with the port number in Arduino C. Check, please.

hajulied, 2022-12-21:
thank you for your efforts. But I don't want to use Arduino. I'm only looking for MPY firmware for the board (RP..) which starts at plugin, or after pressing "run". Then I want to read some sensors (DHT11) and show the values on the display.  Next step is transferring them by WLAN to a Raspberry Pi or to store them in a CSV-file. 
But I'm a beginner and have to solve it by trial and error.
FW Version 1.19.1 from 2022-06-18 seems to be right.

mmMicky, 2022-12-22:
I think Arduino is a great help for beginners. Because it has a library of sensors provided by many community members, and can maximize the performance of the chip. If you really don't want to use ARDUINO. The mpy firmware can try to use the firmware we provide. It has integrated the driver displayed by ST7789, and it can be used only by importing the configuration. For more mpy content, please refer to this document 
https://docs.micropython.org/en/latest/

hajulied, 2023-01-18:
Hi, the board was defective. Today I received a replacement - all works fine.

closed 2023-01-18 by hajulied