Xinyuan-LilyGO/T-PicoC3 #34: Fail to upload with platformio on linux

dchapkine, 2023-03-15:
Hi,

I am failing to upload to the T Pico with platformio (vscode), on linux (ubuntu 22.04)

I use same config as in tutorial

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
board_build.core = earlephilhower
framework = arduino
build_flags = 
	-I ../include
lib_deps = 
	khoih-prog/ESP_AT_Lib@^1.4.1
```

Tryied with this config as well, same result:
 
```
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
```

Compilation works, but not the upload

```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.3% (used 8720 bytes from 262144 bytes)
Flash: [          ]   2.7% (used 56460 bytes from 2093056 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: /dev/ttyS0
Forcing reset using 1200bps open/close on port /dev/ttyS0
Uploading .pio/build/pico/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
======================================================== [FAILED] Took 17.61 seconds ========================================================
```

I am not sure what is going on, but I would say that the board is not detected by the host machine ?

I did boot in 2040 mode (blue lights) via usb C <-> usb A cable

Any Idea of what could be the issue here ?

jimemo, 2023-03-15:
@dchapkine this issue is is solved here:
https://community.platformio.org/t/raspberry-pico-board-first-use-upload-port-search-leads-to-unboundlocalerror/29264/4 
After your build completes: The first time you do your upload, with PIO still open in the background, put the device into boot mode by unplugging the usb, holding the boot button on the device and plugging in the usb again. Now manually upload the newly created firmware.utf file using windows (or whatever) file explorer. Thereafter PIO should detect the COM port properly. If that still doesn't work then try manually specifying your com port in platformio.ini https://docs.platformio.org/en/latest/projectconf/sections/env/options/upload/upload_port.html

dchapkine, 2023-03-27:
Hi, I think the fix works under windows...

But on linux (ubuntu 2204), the device is never detected (even with bootsel pressed)
`sudo dmesg -w` shows nothing... and devices in vscode is empty

Any idea why the device is not detected on linux ?
Does it need a specific driver ?