Xinyuan-LilyGO/T-PicoC3 #35: What does PWR_EN do on GPIO22?

maxpower-electronics, 2023-03-15:
Hello!
I would like to know what does PWR_EN do on GPIO22? How did I understand this display power management?
[image]
[image]

ExperiMentor, 2023-03-15:
It supplies power to the display. High = display on, Low for display off to save battery power. See https://github.com/Bodmer/TFT_eSPI/discussions/1638

maxpower-electronics, 2023-03-15:
Thanks for the answer. But the fact is that I do not give any signal to this pin, but the display turns on for me. I am using CircuitPython and the adafruit_st7789 library.

ExperiMentor, 2023-03-15:
That's because it defaults to ON.
If you turn GPIO22 off, the display should go blank.