Xinyuan-LilyGO/T-PicoC3 #38: Unable to program device -- Fast blink of blue LED

pjpmarques, 2023-08-07:
I've just received three T-PIcoC3 boards, new from factory. I'm unable to program any of them.

When connecting the USB cable, the blue LED blinks quickly. When reversing the cable, the green one blinks. In any case, I'm unable to flash anything to the device. I'm not even able to see a serial/COM port available. Also, the TFT is blank, not showing the factory demo upon connecting.

pjpmarques, 2023-08-10:
Small update: I've tried a number of different USB-C cables (including original Mac USB-C cables). Doesn't work. But when I connect these devices using a cable that has a USB Type-A connector on one side and a USB Type-C connector on the other everything works correctly on all the boards. 

I'm using MacOS 13.5 (Ventura). Not sure if that has anything to do with it.

dcmcshan, 2024-05-15:
I am experiencing this same issue on Somona 14.1.1.  Any updates?

juniorgarcia, 2024-06-13:
Using an USB-A <-> USB-C cable worked for me too... crazy. I don't think that this is a Mac-related because plugin-in on a power bank resulted in the same issue.

owntheweb, 2024-06-18:
To confirm, USB-C to USB-A/B works for me as well on my newer PC. I find this to be the case with multiple microntrollers (I don't recall why). It might be worth a mention in the instructions.